arduino-libraries/WiFiNINA #209: Random disconnection from AP

Matt93-00, 2022-01-21:
Hi people, I am using an Arduino Uno wifi rev 2 with the library WifiNINA. I attach also my code. With this, I can turn on and turn off an imu connected to the board via wifi. When I send the command ON after some variable seconds the device disconnects from AP, forcing the arrest of the functioning of the system. 
With another board using WIFI.h library and the same sketch, all is working properly.
I would like to have help to develop the code of wifinina to have no problem of that kind.
Thank you for the support.

`#include <SPI.h>
#include <WiFiNINA.h>
#include "arduino_secrets.h"
#define  _UART_USB_ Serial

#include <Wire.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_BNO055.h>
#include <utility/imumaths.h>

Adafruit_BNO055 bno = Adafruit_BNO055(55);

float q;
int Value, readacc;

///////please enter your sensitive data in the Secret tab/arduino_secrets.h
char ssid[] = SECRET_SSID;    // your network SSID (name)
char pass[] = SECRET_PASS;    // your network password (use for WPA, or use as key for WEP)
int keyIndex = 0;             // your network key Index number (needed only for WEP)

int status = WL_IDLE_STATUS;
int read_acc = 0;
WiFiServer server(80);

void setup() {

  _UART_USB_.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  /* Initialise the sensor */
  bno.begin();
  delay(1000);
  bno.setExtCrystalUse(true);

  _UART_USB_.println("Access Point Web Server");

  // by default the local IP address of will be 192.168.3.1
  // you can override it with the following:
  // WiFi.config(IPAddress(10, 0, 0, 1));

  if (strlen(pass) < 8) {
    _UART_USB_.println("Creating access point failed");
    _UART_USB_.println("The Wi-Fi password must be at least 8 characters long");
    // don't continue
    while (true);
  }

  // print the network name (SSID);
  _UART_USB_.print("Creating access point named: ");
  _UART_USB_.println(ssid);

  //Create the Access point
  status = WiFi.beginAP(ssid, pass);
  if (status != WL_AP_LISTENING) {
    _UART_USB_.println("Creating access point failed");
    // don't continue
    while (true);
  }

  // wait 10 seconds for connection:
  delay(10000);

  // start the web server on port 80
  server.begin();

  // you're connected now, so print out the status
  printWiFiStatus();

}

void loop() {

  // compare the previous status to the current status
  if (status != WiFi.status()) {
    // it has changed update the variable
    status = WiFi.status();

    if (status == WL_AP_CONNECTED) {
      // a device has connected to the AP
      _UART_USB_.println("Device connected to AP");
    } else {
      // a device has disconnected from the AP, and we are back in listening mode
      _UART_USB_.println("Device disconnected from AP");
    }
  }

  WiFiClient client = server.available();   // listen for incoming clients

  if (client) {                             // if you get a client,
    while (client.connected()) {            // loop while the client's connected

      if (client.available()) {             // if there's bytes to read from the client,
        String currentLine = client.readStringUntil('\n');

        // Check to see if the client request was "GET /.....":
        if (currentLine.indexOf("GET /RON") != -1) {
          sendHeader(client);
          client.println("ON ");
          _UART_USB_.print("ON ");
          _UART_USB_.println(currentLine);
          readacc = 1; // accendo readacc

        } else  if (currentLine.indexOf("GET /ROFF") != -1) {
          sendHeader(client);
          client.println("OFF ");// GET /Lr turns the Red LED off
          _UART_USB_.print("OFF ");
          _UART_USB_.println(currentLine);
          readacc = 0;  //spengo readacc
        }

        // buffer empty
          client.read();
        break;

      } // end if (client.available())

    } // end while (client.connected())
    client.stop();
    _UART_USB_.println("Client disconnesso");
  } // end if (client)
  if ( readacc == 1 ) {

    /* Event code */
    imu::Quaternion quat = bno.getQuat();
    q = quat.w();
    _UART_USB_.println(q);
    delay(100);

  }
}

void printWiFiStatus() {
  // print the SSID of the network you're attached to:
  _UART_USB_.print("SSID: ");
  _UART_USB_.println(WiFi.SSID());

  // print your Wi-Fi shield's IP address:
  IPAddress ip = WiFi.localIP();
  _UART_USB_.print("IP Address: ");
  _UART_USB_.println(ip);

  //print where to go in a browser:
  Serial.print("To see this page in action, open a browser to http://");
  Serial.println(ip);

}

void sendHeader(WiFiClient & locClient) {
  locClient.println("HTTP/1.1 200 OK");
  locClient.println("Content-Type: text/html");
  locClient.println("Connection: close");
  locClient.println();
}`

rbergo, 2023-07-13:
Hello,
I got a similar issue using WifiNina on the MKR1010. The firmware connected well but disconnected a lot of times and the ping time from any PC to the MKR1010 was very slow....
I resolved the issue by calling `Wifi.noLowPowerMode()` before doing the `Wifi.begin()`.
Now it seems very stable and the response time of the Modbus TCP server on MKR1010 has reduced from 400ms to 10ms !!

J1mbo, 2023-07-15:
@Matt93-00 - do you have multiple APs in your environment?

rbergo, 2023-07-17:
Yes, but only one is configured on the SSID used for the test

Matt93-00, 2023-07-19:
Hi all, Yes I have more than 1 AP in my environment. It is curious that Wifi.noLowPowerMode() can solve the problem. I m glad to hear that for You. By the way, I switched to WIfi library, no more WifiNina. Sometimes it lost disconnection again, but for my task, it's enough