arduino-libraries/WiFiNINA #214: Cant connect to eduroam with Arduino Nano 33 IoT

iS3cr3Ti, 2022-03-02:
Hello,
I am currently using the WiFiNINA Library (Sketch "ConnectWithWPA2Enterprise.ino"). I saw that several other Universities got it done with the ESP32. 

These Arguments are needed for the WLAN-Connection:
`status = WiFi.beginEnterprise(ssid, user, pass, identity, caCert);`

But I dont get it working. I received the following login data from my university:

EAP-Method: TTLS or PEAP
Outer Identity with eduroam as identity
Certificate: Needed
Inner Identity with MSCHAPv2 (Username and Password)

Is there a way to specify these settings directly? Or am I doing something wrong?

per1234, 2022-03-02:
Hi @iS3cr3Ti. Thanks for your report.

Please provide a detailed description of the problem you are having, including the full and exact text of any associated error or warning messages.

Please open the Arduino Serial Monitor while the "**ConnectWithWPA2Enterprise**" and the share here the text that is printed.

iS3cr3Ti, 2022-03-02:
The ConnectWithWPA2Enterprise Code:

`/*
  This example connects to a WPA2 Enterprise WiFi network.
  Then it prints the MAC address of the WiFi module,
  the IP address obtained, and other network details.

  Based on ConnectWithWPA.ino by dlf (Metodo2 srl) and Tom Igoe
*/
#include <SPI.h>
#include <WiFiNINA.h>
#include <WiFi.h>

#include "arduino_secrets.h"
///////please enter your sensitive data in the Secret tab/arduino_secrets.h
char ssid[] = SECRET_SSID;  // your WPA2 enterprise network SSID (name)
char user[] = SECRET_USER;  // your WPA2 enterprise username (Inner Method)
char pass[] = SECRET_PASS;  // your WPA2 enterprise password (Inner Method)
char identity[] = SECRET_IDENTITY;  //your WPA2 enterpise identity (Outer Method)
char caCert[] = SECRET_CERT;  //your WPA2 enterprise certificate
int status = WL_IDLE_STATUS;     // the WiFi radio's status
byte mac[6];

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  // check for the WiFi module:
  if (WiFi.status() == WL_NO_MODULE) {
    Serial.println("Communication with WiFi module failed!");
    // don't continue
    while (true);
  }

  String fv = WiFi.firmwareVersion();
  if (fv < WIFI_FIRMWARE_LATEST_VERSION) {
    Serial.println("Please upgrade the firmware");
  }

  // attempt to connect to WiFi network:
  while (status != WL_CONNECTED) {
    Serial.print("Attempting to connect to WPA SSID: ");
    Serial.println(ssid);
    // Connect to WPA2 enterprise network:
    // - You can optionally provide additional identity and CA cert (string) parameters if your network requires them:
    //      WiFi.beginEnterprise(ssid, user, pass, identity, caCert)
    status = WiFi.beginEnterprise(ssid, user, pass);
    Serial.print("This is the Status: ");
    Serial.println(status);
    Serial.print("This is the Idle Status: ");
    Serial.println(WL_IDLE_STATUS);
    Serial.print("This is the Wanted Status: ");
    Serial.println(WL_CONNECTED);

    // wait 10 seconds for connection:
    delay(10000);
  }`

The arduino_secrets.h:
`#define SECRET_SSID "eduroam"
#define SECRET_IDENTITY "Outer Identity"
#define SECRET_USER "Inner Method Username"
#define SECRET_PASS "Password for the User"
#define SECRET_CERT "-----BEGIN CERTIFICATE-----\n" \
"MIIDwzCCAqugAwIBAgIBATANBgkqhkiG9w0BAQsFADCBgjELMAkGA1UEBhMCREUx\n" \
*********
"BSeOE6Fuwg==\n" \
"-----END CERTIFICATE-----"`

Output of the Serial Monitor:
`12:34:54.399 -> Attempting to connect to WPA SSID: eduroam
12:35:44.612 -> This is the Status: 1
12:35:44.612 -> This is the Idle Status: 0
12:35:44.612 -> This is the Wanted Status: 3
12:35:54.611 -> Attempting to connect to WPA SSID: eduroam
12:35:54.744 -> This is the Status: 6 (WRONG_PASSWORD)
12:35:54.744 -> This is the Idle Status: 0
12:35:54.744 -> This is the Wanted Status: 3
12:36:04.741 -> Attempting to connect to WPA SSID: eduroam
12:36:06.867 -> This is the Status: 4
12:36:06.867 -> This is the Idle Status: 0
12:36:06.867 -> This is the Wanted Status: 3
`

So this is the Code, which I am currently using. There is no Error Message or Warning Message. The Password is working, because I am using it with my Phone too. I just cant connect to the network.

iS3cr3Ti, 2022-03-07:
Should I provide more Informations or is there anything I can do?

ChathuZ, 2023-01-12:
> Should I provide more Informations or is there anything I can do?

Did you find a solution for this?