arduino-libraries/WiFiNINA #223: SSL Certificate Updater

JEG8, 2022-07-08:
Can anyone provide the code used to update the SSL certificates on the NINA? Is there any independent sketch to update SSL certificates? I would like to do this remotely via a server, instead of having to plug into the Arduino every time (its not practical in a product). Also, is the crypto chip (ATECC608A) used to store the certificate or is it just loaded directly onto the NINA?

per1234, 2022-07-08:
Hi @JEG8. Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we will be able to help you out over there:

https://forum.arduino.cc/

closed 2022-07-08 by per1234

JEG8, 2022-07-08:
Hi,

Thank you for pointing my thread in the right direction. I still think this post is applicable for a new feature request. Is there another way to update the SSL certificates without having to update via the arduino IDE updater tool?

Thank you