arduino-libraries/WiFiNINA #226: client.available() return 0 if no client.read() or equivalent used before

Gouttfi, 2022-09-09:
Hi, I'll try to correctly explain the problem in english but i'm french.

I have a script to get data from webpage, and I need only the body, so I made this code to separate header and body : 

``` 
while (client.available()) {

  // Collect hearders, row by row
  header = client.readStringUntil('\n');

  // If the row is \r, the next row will be the body
  if (header == "\r") {
    while (client.available()) {

      // Collect body, row by row
      String body = client.readStringUntil('\n');
      // My own treatment for body response (the response is only one row)
    }
  }
}
```

This code is supposed to work like this : 
While data is available, read until \n, store it to "header" variable. If header == "\r", so the headers included in the http response are all read. So I can now read and store the body.
BUT, the first client.available() return always false, so the code are never triggered.

The solution I found is reading the first header line by adding this line at the top of my code, before the first _while(client.available())_ : 

`    String header = client.readStringUntil('\n'); `

This start retrieving available information without knowing if it is really available. The "header" var return correctly the first header sent by HTTP response, and next, if I do client.available(), it return correctly the number of bytes available for reading.

To resume, client.available() return 0 if nothing wasn't read before, is it really a bug or misinterpretation of the function available() ?

Thanks

JAndrassy, 2022-09-09:
it takes time until the response arrives so the first few calls to `available()` return 0.
`readStringUntil` is one of the blocking functions. it waits for the next character until timeout. default timeout is 1 second.

Gouttfi, 2022-09-09:
Thank you for the answer :)

So how can I modify this function correctly without doing this improvised patch ? It seems to work perfectly but it seems trivial !

JAndrassy, 2022-09-09:
maybe don't check available() at all until you don't get to http body.

if the response doesn't arrive at all, the readStringUntil will timeout and then avaliable() will be zero and the sketch continues. you should add a check if the returned String isn't empty.

if a response is received, it is HTTP so it has at least one line. you have to process the line to know the HTTP response code. then you can skip the headers (` client.find((char*) "\r\n\r\n");` will do) and read the body.

but maybe you want to use the ArduinoHTTPClient library

Gouttfi, 2022-09-10:
Hi, thanks for your help.

Finally, I applied your suggestion and replaced all my code to only this :

```
    client.find((char*) "\r\n\r\n");
    String header = client.readStringUntil('\n');
    String body = client.readStringUntil('\n');
```

Seems to work correctly

closed 2022-09-10 by per1234