arduino-libraries/WiFiNINA #228: WiFi Web Server example not compatible on Windows & Mac using WiFiNINA

jazhe, 2022-09-15:
Hello there!

I'm using the Nano RP2040 example and WiFi Web Server (from the WiFiNINA library) on Windows (my main computer), even though it compiles and loads fine, it can't establish a connection through the browser. When I try it on Linux everything works fine. I also tried it on Mac, but it's the same problem as on Windows.
Also, if I try to connect it using Firefox, there is a delay when it is loading the WebServer, but in Chrome it works perfectly.
I would like to know why this happens and also if could be a possible solution.

Best,
Jaz

khoih-prog, 2022-09-15:
Hi @jazhe 

Thanks for your interest and usage of the library.

> it can't establish a connection through the browser. When I try it on **Linux everything works fine**. I also tried it on **Mac**, but it's the same problem as on **Windows**.

> but in **Chrome it works perfectly**.

That means there must be something **wrong** with your **browsers** and/or **Mac and Windows.**

> I would like to know why this happens and also if could be a possible solution.

I suggest you post the issue on **related** Forums to have the issue fixed. Tell them try to **imitate** and be **compatible** to `Chrome on Linux`.

I'm sure I don't have time to spend on their issues.

Good Luck,

JAndrassy, 2022-09-21:
you write "When I try it on Linux ..". try what: build and upload the sketch or connect from browser?

jazhe, 2022-09-21:
Sorry, my mistake. 
I mean, when I try to connect from the browser. Using Linux and Chrome, it works well and quickly. But when I try the same process using Windows and Chrome, it keeps loading, but nothing happens.

sabas1080, 2023-03-02:
WIFINiNA: v1.8.13
Core mbed: v3.4.1

working windows and linux