arduino-libraries/WiFiNINA #232: WifiNina cannot cope with Business wifi environment

Russell108, 2022-11-24:
Ability to connect to a specific Access point where many access points share the same SSID
Currently, in Business hotel environment where the same SSID password is used across a set of access points 

The Arduino MKR1010 and the Arduino IOT 33 do not connect to a sensible access point instead connect to one with a real signal.

Ideally, there would be. A function in the library to select the best access point by signal strength for a given SSID

per1234, 2022-11-24:
Hi @Russell108. Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino-libraries/WiFiNINA/issues/200.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2022-11-24 by per1234

Russell108, 2022-11-24:
Is this issue going to be resolved