arduino-libraries/WiFiNINA #245: setHostname not working with beginAP

Sitebulb, 2023-05-03:
setHostname works fine when connecting to an existing access point, but it's not working when creating one. 

Working

`WiFi.setHostname(hostname);
...
while (status != WL_CONNECTED) {
        status = WiFi.begin(ssid, pass);
        delay(10000);
 }`

Not working

`WiFi.setHostname(hostname);
 WiFi.config(APIP, APIP, APIP, IPAddress(255, 255, 255, 0));
 status = WiFi.beginAP(ssid, pass, apChannel);`

Is this by design, or not possible?

JAndrassy, 2023-05-03:
where do you expect to see/use the hostname?

Sitebulb, 2023-05-03:
-- Please reply above this line --

Hello Arduino-Libraries/wifinina,

 Thanks for contacting Sitebulb support. If you are having a problem
with an audit, report, or have found a bug, please can you reply to
this email with the following:

 	* The URL of the website you are auditing.
 	* Whether you are experiencing the same issue across all/multiple
audits or only on this one
 	* An export of the audit itself (this does not mean Excel
spreadsheets, it means the .sbp audit file which you can export and
share). Check this guide - how to share a Sitebulb audit [1].
 	* Your error log files (find these here [2]).
 	* Screenshots (or even better, screencasts!) of the issue. You can
use something like TechSmith Capture [3] to quickly create these and
upload for free.

 Please can you also confirm this issue exists with latest version of
Sitebulb? If you are not on the latest version, you can grab it here
[4].

 If you just need help using Sitebulb (e.g. getting started, how do I,
billing etc...) you may be able to find your answer quickly from the
resources on our website:

 	* Learn Sitebulb [5] - a dedicated area for learning how to make the
most of the Sitebulb tool.
 	* Documentation [6] - this includes explanations about setting up
and running audits, how audits and projects work, and
licensing/subscriptions/billing.
 	* Guides [7] - various how-to guides, which will help in some common
scenarios, such as 'How to crawl JavaScript websites [8]'.
 	* Feature Requests [9] - use this dedicated area to add your own
feature requests and get votes from the community, and vote or comment
on other suggestions.

 We aim to answer all queries within 24 hours, during working hours:
Mon-Fri, 9am-5:30pm (UK time).

 Thank you!

 The Sitebulb Team

Links:
------
[1]
https://sitebulb.com/documentation/audits-projects/how-to-share-an-audit/
[2]
https://sitebulb.com/documentation/troubleshooting/help-sitebulb-has-errored-or-crashed/#error-logs
[3]
https://www.techsmith.com/learn/tutorials/capture-desktop/capture-app/
[4] https://sitebulb.com/release-notes/
[5] https://sitebulb.com/learn-sitebulb/
[6] https://sitebulb.com/documentation/
[7] https://sitebulb.com/resources/guides/
[8]
https://sitebulb.com/resources/guides/how-to-crawl-javascript-websites/
[9] https://sitebulb.featureupvote.com/

garethbrown, 2023-05-03:
@JAndrassy want my nano rp2040 connect to be the access point and view a HTML page via the hostname, with having to know the IP address. The device is going to be used in an environment where there is no Wifi router.

JAndrassy, 2023-05-04:
so I am not an expert on this, but in my understanding, hostname is sent with the DHCP request and then may be used by the router/AP with mDNS.
if you want hostname resolution on SoftAP of the MKR, you will have to start a DNS or mDNS server. I don't know if there is a library for mDNS or DNS over an Arduino WiFi library.